younghyunjo/esp32-homekit #6: mongoose.c:9026:3: note: in expansion of macro 'sigemptyset'

EvanLyu, 2018-04-02:
I cloned a new repository.
I have some compilation errors on mongoose.c
My operating system is MacOS 10.13.3 and uses the latest ESP-IDF / esp32-homekit
What steps have I made wrong?

This error occurs the first time I use it, but it will not happen afterwards.
[image]

[image]
[image]

younghyunjo, 2018-04-02:
I am working with ubuntu16.04, lastest ESP_IDF and xtensa-esp32-elf-1.22.0-75 toolchain.
```
commit aca000869406e35eba0d5fd7064304ba9e5acead
Merge: 8eaae96 d6b1d0b
Author: Ivan Grokhotkov <ivan@espressif.com>
Date:   Mon Apr 2 13:48:46 2018 +0800

    Merge branch 'bugfix/sdspi_init_v1_card' into 'master'
    
    sdspi: compatibility fixes for SD v1.0 cards
    
    See merge request idf/esp-idf!1927

```
I am not sure the exact reason 
but I think the reason seems like toolchain
please check your toolchain version

EvanLyu, 2018-04-03:
Thanks.
I downgrade toolchain to xtensa-esp32-elf-1.22.0-75 and upgrade mongoose to 6.11 .
Build success on my mac 10.13.3

closed 2018-04-03 by EvanLyu

younghyunjo, 2018-04-03:
I upgraded mongoose to 6.11 in master branch

jjoe64, 2018-06-06:
I had the same error and I just had to run `make all` to fix it