younghyunjo/esp32-homekit #7: Make error

ysdndev, 2018-04-03:
Following build instructions. On make command:

Makefile:8: /make/project.mk: No such file or directory
make: *** No rule to make target `/make/project.mk'.  Stop.

Did I miss something?

younghyunjo, 2018-04-03:
Did you install esp-idf?
https://github.com/espressif/esp-idf

ysdndev, 2018-04-03:
Got esp-idf installed configured and woking, but came across this build error:

AR build/app_update/libapp_update.a
CC build/arduino/cores/esp32/libb64/cdecode.o
make[1]: *** No rule to make target `cores/esp32/libb64/cencode.o', needed by `libarduino.a'.  Stop.
make: *** [component-arduino-build] Error 2

Any help would be appreciated.

younghyunjo, 2018-04-08:
@ysdndev 
Everything works fine?
If it works well, please close the issue

ysdndev, 2018-04-08:
Sorry, haven't had a chance to test. Close for now. Thanks!

closed 2018-04-08 by ysdndev