younghyunjo/esp32-homekit #10: Hang-up while pairing

younghyunjo, 2018-04-21:
This issue reported by @streicherlouw

---

Hi.

I am trying this code for the first time today, and having a bit of trouble with pairing. The example SWITCH object also shows up on my homekit network, and I can pair with it, but in the last step of the pairing, something seems to go wrong with the status reporting.

The last step in pairing is where the ipad/iphone retrieves the status of the newly added accessory. In that step, my ipad just never gets a response and eventually times out. I can subsequently quit the ipad app and restart the ESP32, and then tings work fine, but the pairing always ends in a status read failure for me. Does yours pair successfully?

Pairing log:

MONITOR
--- idf_monitor on /dev/cu.usbserial-DN027YZF 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun 8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5656
load:0x40078000,len:0
ho 12 tail 0 room 4
load:0x40078000,len:14008
entry 0x4007860c
I (31) boot: ESP-IDF v3.1-dev-826-g4e982d4b 2nd stage bootloader
I (31) boot: compile time 11:21:39
I (42) boot: Enabling RNG early entropy source...
I (42) boot: SPI Speed : 40MHz
I (42) boot: SPI Mode : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label Usage Type ST Offset Length
I (60) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (67) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (75) boot: 2 factory factory app 00 00 00010000 00100000
I (82) boot: End of partition table
I (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1be48 (114248) map
I (135) esp_image: segment 1: paddr=0x0002be70 vaddr=0x3ffb0000 size=0x03790 ( 14224) load
I (141) esp_image: segment 2: paddr=0x0002f608 vaddr=0x40080000 size=0x00400 ( 1024) load
0x40080000: _iram_start at /Users/streicher/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (143) esp_image: segment 3: paddr=0x0002fa10 vaddr=0x40080400 size=0x00600 ( 1536) load
I (152) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x81638 (529976) map
0x400d0018: _stext at ??:?

I (345) esp_image: segment 5: paddr=0x000b1658 vaddr=0x40080a00 size=0x0ec80 ( 60544) load
I (371) esp_image: segment 6: paddr=0x000c02e0 vaddr=0x400c0000 size=0x00000 ( 0) load
I (380) boot: Loaded app from partition at offset 0x10000
I (381) boot: Disabling RNG early entropy source...
I (383) cpu_start: Pro cpu up.
I (386) cpu_start: Starting app cpu, entry point is 0x40080f74
0x40080f74: call_start_cpu1 at /Users/streicher/esp/esp-idf/components/esp32/cpu_start.c:224

I (0) cpu_start: App cpu up.
I (397) heap_init: Initializing. RAM available for dynamic allocation:
I (404) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (410) heap_init: At 3FFB9A38 len 000265C8 (153 KiB): DRAM
I (416) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (422) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (429) heap_init: At 4008F680 len 00010980 (66 KiB): IRAM
I (435) cpu_start: Pro cpu start user code
I (117) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (261) wifi: wifi firmware version: c09b606
[... 189 lines cut ...]
[HTTPD] MG_EV_RECV
[INFO] HTTP request from 10.0.1.179:53399: POST /pairings
[PAIRINGS] STATE:1 METHOD:3
[HAP][INFO] _decrypt end 216
[HTTPD] Connection 0x3ffc9f74 from 10.0.1.158:56759
[HTTPD] MG_EV_RECV
[INFO] HTTP request from 10.0.1.158:56759: POST /pair-verify
[HTTPD] DEFAULT:102
[HTTPD] DEFAULT:100
[HTTPD] MG_EV_SEND. 0
[HTTPD] MG_EV_RECV
[INFO] HTTP request from 10.0.1.158:56759: POST /pair-verify
[HTTPD] DEFAULT:102
[HTTPD] DEFAULT:100
[HTTPD] MG_EV_SEND. 1
[HTTPD] MG_EV_RECV
[INFO] HTTP request from 10.0.1.158:56759: GET /accessories
[HAP][INFO] _decrypt end 77
[HTTPD] MG_EV_SEND. 1
[HTTPD] MG_EV_RECV
[INFO] HTTP request from 10.0.1.158:56759: POST /pairings
[PAIRINGS] STATE:1 METHOD:3
[HAP][INFO] _decrypt end 216
[HTTPD] MG_EV_CLOSEConnection 0x3ffcade0 closed
[HTTPD] MG_EV_CLOSEConnection 0x3ffca960 closed

younghyunjo, 2018-04-21:
@streicherlouw
When i read your log, Multiple iOS devices are trying to paring at the same time.

Http Request IPs
```
10.0.1.179  
10.0.1.158  
10.0.1.160:  
10.0.1.177  10.0.1.178 10.0.1.162   10.0.1.179 10.0.1.158
```

streicherlouw, 2018-04-22:
Hi younghyunjo

I am only pairing with one device, but has my home set up with two iPads and two iPhones (for two people) controlling the same HomeKit devices. Once I pair the ESP32 on one device, the pairing is automatically synchronised to the other devices, and I suspect that those devices are requesting status information from the ESP32. As most HomeKit networks will always have the a gateway (like an Apple TV) in addition to the phones or iPads controlling the devices, it would be a great advancement in the practical usability of the code if it could cope with multiple devices.

younghyunjo, 2018-04-23:
@streicherlouw 

Thanks for your reply.
If I finish to add the feature, I will let you know.
Thanks for using my code.  :)

streicherlouw, 2018-04-23:
Thank you very much. I look forward to trying the next release very much as I have many projects that I want to integrate with HomeKit one the ESP32. Please let me know if I can help in any way with testing. I would be happy to try and debug as often as you require.

younghyunjo, 2018-04-25:
@streicherlouw 

I implement "Add pairings" feature for multiple iOS devices.
But I can`t test it, because I only have one Iphone.
Would you test the code for me?
you can download modified source code at "feature/support-multi-devices" [branch](https://github.com/younghyunjo/esp32-homekit/tree/feature/support-multi-devices)

If it`s not works well. please attach serial log.

thanks you

streicherlouw, 2018-04-25:
Hi .

Thank you for implementing the feature so quickly. A full log of the compile and running of the code is at the end of this message. I include the compilation log as there were a few warnings that may or may not help you.

I have been able to test the new code, and can confirm that the primary pairing works now, but that the new code does not allow any of the additional devices to actually control the ESP32.

Here is what my network looks like: There are 6 iOs devices. Five run IOS11, one runs iOS 9. Their ip address all start with 10.0.1. and end with:
.160 >> iPhone6
.158 >> iPad Air2 (This device also acts as the homekit hub (see https://support.apple.com/en-au/HT207057)
.179 >> iPad 6
.162 >> iPad Pro
.178 >> iPhone X
.177 >> iPad Mini

The situation is as follows:
I am pairing from the iPhone 6 (.160). This works perfectly now (Thanks! It did not work before). Then homekit automatically syncs the new SWTICH to all of the other iOS devices. These devices seem to contact the SWITCH to pair. Then I open the "home" app on the iPad 6 (.179) to see if I can control it from there, but it fails with a message "No Response". Then I try to control the SWITCH with with iPad 6 (.179), but it just keeps saying "No response".

I think the problem may be that the pairings to the other devices may not be completing successfully. The thing that makes me this so is that when I switch off my iPhone 6's wifi  to force homekit to update over the 4G connection through the homekit hub (.158), nothing shows up in the ESP32 monitor. This would suggest that the homekit hub does not know the SWITCH or is not paired with it correctly.

One more really strange thing is that when I subsequently remove the SWITCH from the iPhone 6 and add it to the iPad 6, it acts as you would expect, now the iPad can control the SWITCH. BUT, in this configuration, the iPhone 6 can now *also* control the SWITCH. It looks like the SWITCH or the iOS devices remembers something about the pairing. But even with this strange both-working configuration, it is not quite right. If you switch the output on or off at the paired device (the iPad in this case) the status on the iPhone changes too, but if you do the change on the iPhone, the iPad does not update. Also, when you do this, sooner or later the ESP32 throws an exception that looks like this:

[MAIN] LED WRITE. 1
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x40115afe  PS      : 0x00060e30  A0      : 0x801133ae  A1      : 0x3ffc71a0  
0x40115afe: mg_send at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3119

A2      : 0x4bc00062  A3      : 0x3ffcc5f8  A4      : 0x00000097  A5      : 0x00000002  
A6      : 0x3ffcaedc  A7      : 0x00000085  A8      : 0x00810000  A9      : 0x00810000  
A10     : 0x00000a3e  A11     : 0x40a47de9  A12     : 0x0000000c  A13     : 0x00000a3e  
A14     : 0x7ff00000  A15     : 0x7ff47de9  SAR     : 0x00000014  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00810008  LBEG    : 0x400e4320  LEND    : 0x400e433f  LCOUNT  : 0x00000000  
0x400e4320: __divdf3 at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:1454

0x400e433f: __divdf3 at /Volumes/build/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/config/xtensa/ieee754-df.S:1481

Backtrace: 0x40115afe:0x3ffc71a0 0x401133ab:0x3ffc71c0 0x40113764:0x3ffc7210 0x400d25ad:0x3ffc7240 0x4011e24a:0x3ffc7260 0x4011362c:0x3ffc72a0 0x40113708:0x3ffc7480 0x40113a25:0x3ffc74b0 0x401164ba:0x3ffc74f0 0x40116a92:0x3ffc7530 0x40116ea9:0x3ffc75e0 0x401164ba:0x3ffc7780 0x40116fca:0x3ffc77c0 0x4011731d:0x3ffc77f0 0x40117603:0x3ffc7860 0x40115a66:0x3ffc78b0 0x40113a76:0x3ffc78d0
0x40115afe: mg_send at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3119

0x401133ab: encrypt_send at /Users/streicher/esp/esp32-homekit/src/hap.c:452

0x40113764: hap_event_response at /Users/streicher/esp/esp32-homekit/src/hap.c:452

0x400d25ad: led_write at /Users/streicher/esp/esp32-homekit/examples/switch/main/main.c:68

0x4011e24a: hap_acc_characteristic_put at /Users/streicher/esp/esp32-homekit/src/accessories.c:340

0x4011362c: _plain_msg_recv at /Users/streicher/esp/esp32-homekit/src/hap.c:452

0x40113708: _encrypted_msg_recv at /Users/streicher/esp/esp32-homekit/src/hap.c:452
 (inlined by) _msg_recv at /Users/streicher/esp/esp32-homekit/src/hap.c:302

0x40113a25: mg_ev_handler at /Users/streicher/esp/esp32-homekit/src/httpd.c:38

0x401164ba: mg_call at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3119

0x40116a92: mg_http_handler2 at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3119

0x40116ea9: mg_http_handler at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3119

0x401164ba: mg_call at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3119

0x40116fca: mg_recv_common at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3119

0x4011731d: mg_if_recv_tcp_cb at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3119
 (inlined by) mg_handle_tcp_read at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3554
 (inlined by) mg_mgr_handle_conn at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3679

0x40117603: mg_socket_if_poll at /Users/streicher/esp/esp32-homekit/src/mongoose.c:3119
[... 1024 lines cut ...]
[HTTPD] DEFAULT:102
[HTTPD] DEFAULT:100
[HTTPD] MG_EV_SEND. 0
[HTTPD] MG_EV_CLOSEConnection 0x3ffc9c24 closed
[HTTPD] Connection 0x3ffc9c24 from 10.0.1.179:57490
[HTTPD] MG_EV_RECV
[... cut ...]
[HTTPD] MG_EV_CLOSEConnection 0x3ffc9c24 closed
[HTTPD] Connection 0x3ffc9c24 from 10.0.1.158:59638
[HTTPD] MG_EV_RECV
[INFO] HTTP request from 10.0.1.158:59638: POST /pair-verify
[HTTPD] DEFAULT:102
[HTTPD] DEFAULT:100
[HTTPD] MG_EV_SEND. 0
[HTTPD] MG_EV_CLOSEConnection 0x3ffc9c24 closed
[HTTPD] Connection 0x3ffc9c24 from 10.0.1.158:59639
[HTTPD] MG_EV_RECV
[INFO] HTTP request from 10.0.1.158:59639: POST /pair-verify
[HTTPD] DEFAULT:102
[HTTPD] DEFAULT:100
[HTTPD] MG_EV_SEND. 0
[HTTPD] MG_EV_CLOSEConnection 0x3ffc9c24 closed

younghyunjo, 2018-04-25:
@streicherlouw 

Thanks for test my code and detail explanation.
Please understand me, this is a little tough problem because I can`t reproduce the problem on my desk.

As your comment, there are three problems.

1. Pairing from iPhone6.  "No response" at iPad 6
2. Pairing from iPad 6. The accessory status is not synced when controlled by iPhone6.
3. Pairing from iPad 6, ESP32 throw an exception while control accessory.

I think I can handle No2, No3 problem, but No1 problem is a little strange.
I will dig more about it.

I have 1 question, the every ios devices are using same Apple ID?

streicherlouw, 2018-04-25:
Problem 3 happens only when I turn the accessory on and off on both iPad 6 and iPhone 6 quickly. If I do it slowly only the status problem happens, but no reboot. Problem 3 could be a race condition. 

To answer your question, 4 of the 6 devices have the same apple ID, 2 of the devices (iPhone X and iPad Pro) have a different apple ID that has been invited to the network in the Home app as described on https://support.apple.com/en-au/HT204893. I do not think the apple id makes a difference, but can check for you by removing the second person. Let me know if you want me to test this.

streicherlouw, 2018-04-30:
As the pairing problem seems to be solved, it might be good to open a new issue for "multiple iOS device access" in stead of pairing hang-up. The pairing works fine now, but I am unable to get a paired device to reliably work from other iOS devices in the homekit group. This spells trouble for both external access as well as automations (where it will be the homekit hub sending the automation commands)

younghyunjo, 2018-05-03:
@streicherlouw 

I was off last few days.
I am enhancing stability now.

I close this issue and open another issue related to stability

closed 2018-05-03 by younghyunjo