younghyunjo/esp32-homekit #11: Compilation error

Laxiyan, 2018-04-24:
Hello,

I follow the steps，

$ git clone  https://github.com/younghyunjo/esp32-homekit.git
$ cd esp32-homekit
$ git submodule update --init --recursive
$ cd examples/switch
$ make
$ make flash

The following error occurred，

esp8266@esp8266-VirtualBox:~/Desktop/esp/esp32-homekit/examples/switch$ make
CC build/esp32-homekit/src/advertise.o
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c: In function '_service_txt_set':
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:36:5: error: unknown type name 'mdns_txt_item_t'
     mdns_txt_item_t hap_service_txt[] = {
     ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:37:9: warning: braces around scalar initializer
         {"c#", adv->service_txt_c_sharp},
         ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:37:9: note: (near initialization for 'hap_service_txt[0]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:37:10: warning: initialization makes integer from pointer without a cast [-Wint-conversion]
         {"c#", adv->service_txt_c_sharp},
          ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:37:10: note: (near initialization for 'hap_service_txt[0]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:37:16: warning: excess elements in scalar initializer
         {"c#", adv->service_txt_c_sharp},
                ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:37:16: note: (near initialization for 'hap_service_txt[0]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:38:9: warning: braces around scalar initializer
         {"ff", "0"},
         ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:38:9: note: (near initialization for 'hap_service_txt[1]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:38:10: warning: initialization makes integer from pointer without a cast [-Wint-conversion]
         {"ff", "0"},
          ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:38:10: note: (near initialization for 'hap_service_txt[1]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:38:16: warning: excess elements in scalar initializer
         {"ff", "0"},
                ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:38:16: note: (near initialization for 'hap_service_txt[1]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:39:9: warning: braces around scalar initializer
         {"pv", "1.0"},
         ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:39:9: note: (near initialization for 'hap_service_txt[2]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:39:10: warning: initialization makes integer from pointer without a cast [-Wint-conversion]
         {"pv", "1.0"},
          ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:39:10: note: (near initialization for 'hap_service_txt[2]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:39:16: warning: excess elements in scalar initializer
         {"pv", "1.0"},
                ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:39:16: note: (near initialization for 'hap_service_txt[2]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:40:9: warning: braces around scalar initializer
         {"id", adv->id},
         ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:40:9: note: (near initialization for 'hap_service_txt[3]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:40:10: warning: initialization makes integer from pointer without a cast [-Wint-conversion]
         {"id", adv->id},
          ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:40:10: note: (near initialization for 'hap_service_txt[3]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:40:16: warning: excess elements in scalar initializer
         {"id", adv->id},
                ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:40:16: note: (near initialization for 'hap_service_txt[3]')
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:41:9: warning: braces around scalar initializer
         {"md", adv->name},
         ^
[... 75 lines cut ...]
 esp_err_t mdns_init(tcpip_adapter_if_t tcpip_if, mdns_server_t ** server);
           ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:98:5: warning: implicit declaration of function 'mdns_hostname_set' [-Wimplicit-function-declaration]
     mdns_hostname_set(host);
     ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:99:5: warning: implicit declaration of function 'mdns_instance_name_set' [-Wimplicit-function-declaration]
     mdns_instance_name_set(name);
     ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:100:22: warning: passing argument 1 of 'mdns_service_add' from incompatible pointer type [-Wincompatible-pointer-types]
     mdns_service_add(name, HAP_SERVICE, HAP_PROTO, port, NULL, 0);
                      ^
In file included from /home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:5:0:
/home/esp8266/Desktop/esp/esp-idf/components/mdns/include/mdns.h:108:11: note: expected 'mdns_server_t * {aka struct mdns_server_s *}' but argument is of type 'char *'
 esp_err_t mdns_service_add(mdns_server_t * server, const char * service, const 
           ^
/home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:100:5: error: too many arguments to function 'mdns_service_add'
     mdns_service_add(name, HAP_SERVICE, HAP_PROTO, port, NULL, 0);
     ^
In file included from /home/esp8266/Desktop/esp/esp32-homekit/src/advertise.c:5:0:
/home/esp8266/Desktop/esp/esp-idf/components/mdns/include/mdns.h:108:11: note: declared here
 esp_err_t mdns_service_add(mdns_server_t * server, const char * service, const 
           ^
make[1]: *** [src/advertise.o] Error 1
make: *** [component-esp32-homekit-build] Error 2
esp8266@esp8266-VirtualBox:~/Desktop/esp/esp32-homekit/examples/switch$

Laxiyan, 2018-04-24:
I upgraded the esp-idf to the latest version and the compile passed. 
thank you.

younghyunjo, 2018-04-24:
That`s good.
Thanks for using my codes. :)

closed 2018-04-24 by younghyunjo