younghyunjo/esp32-homekit #12: temperature sensor ?

Kristian8606, 2018-04-25:
Congratulations on the good project!
I'm not a good programmer yet, but I've done yours project and I really enjoyed it.
I hope we will soon see a project like a temperature sensor or a colored strip.
Excuse me for my bad English.

younghyunjo, 2018-04-27:
@Kristian8606 

Thanks for have a interests about my code.
I ordered temperature, humidity sensor(DHT22) at Aliexpress.
It`s delivering now.

After I stabilize my code, i have a plan to make a temperature example.
Please watch the enhancement of my code, and wait until I make it.

Thanks :)

Kristian8606, 2018-04-29:
I'm not sure my question is if it's about this topic but I will ask it here.
To esp32 i have attached an LED to lights at night, but in the morning it is off.
I'll try to monitor it with a monitor to see why it's off.
What can be the problem, do you have such case?

closed 2018-05-02 by younghyunjo