younghyunjo/esp32-homekit #17: Float values for certain characteristics

TAKeanice, 2018-05-07:
Dear Younghyunjo,

after my first experiments looked good, I went on to adapt the example to my needs. However, I encountered the problem, that it is impossible to hand over float values in the same way you hand over the boolean value for the on/off state of the LED.

What you are using is a `void` pointer, or a cast to it. This is only possible for integer types, and throws an error for floating point types.

I saw, that in your commit  https://github.com/younghyunjo/esp32-homekit/commit/f17edcbfa60bbb5660ddffb7a6a2cfa1c724dbd7 yesterday, you changed the type handling of different characteristics, so you seem to be aware that different characteristics work on different types. But how can we retrieve and return those (especially floating point) values in the callbacks?

younghyunjo, 2018-05-08:
To use floating point values, multiply 100 with sensor value and return as integer.
For example, if you want send the value 36.5, return 3650 in you callback function.

I am sorry for poor documentation and thanks for using my library.

TAKeanice, 2018-05-08:
Thanks for the prompt response and for creating the library in the first place! As for documentation, it would totally suffice to have some comments in the `hap.h`. For example, I am not sure what the `ev_handle` is about. A also suspect the `arg` argument is there for passing a flag or something like that?

TAKeanice, 2018-05-08:
@younghyunjo, sorry to bother you again. It is still about the additional charakteristics that I want to add to the example. Since the `arg` did not work as I expected, I just phased the different characteristics responses into different callbacks. Please excuse the ugly switch statements that I just retained out of lazyness. Anyway, as soon as I comment in lines 341 to 343 [in my branch version of the example](https://github.com/TAKeanice/esp32-homekit/tree/with_sk6812), it becomes impossible to add the device in my home app. I tried to look into the code of your library, but as far as I could see, everything was fine... Could you maybe try my modified example and tell me if it works for you or what is wrong about it? I know it is somewhat too much to ask, but I hope anyway that you consider it.

younghyunjo, 2018-05-09:
@TAKeanice 
I fixed floating point problem, and arg problem.

https://github.com/younghyunjo/esp32-homekit/commit/476a1eac39c2972c917f51a6d760830c7a9921ba
https://github.com/younghyunjo/esp32-homekit/commit/1c246f8657c37547880b91561b53fd5fcb7e3311

younghyunjo, 2018-05-09:
As I comment in lines 341 to 343, It works well.

https://drive.google.com/file/d/1kAldNertFukrp5i2rMC_yigihE4eoVOz/view?usp=sharing
https://drive.google.com/file/d/1F1Mi6gKW4DaBRgH8604hHFQF2AZoOlhN/view?usp=sharing

TAKeanice, 2018-05-09:
Great! My Home app now recognized the device and I can set colors. However, the incoming values don´t seem to get multiplied by 100. I can work around this for now, and I will try to identify where the values come from in your library and propose where to change that.

younghyunjo, 2018-05-09:
@TAKeanice 

Here is example using floating point value.
https://github.com/younghyunjo/esp32-homekit/blob/master/examples/thermometer-hygrometer-dht22/main/main.cpp

TAKeanice, 2018-05-09:
Yes, thank you, I have checked if my code is the same. One part that is missing in your example is where values are written (like in led_write in the other example). There, values come unmultiplied. As an example, see line 202 of my working example (same branch as before).

younghyunjo, 2018-05-09:
@TAKeanice 

Oh! I forgot to handle it in writing situation.
I will fix it later.
Thanks.

closed 2020-10-01 by TAKeanice