younghyunjo/esp32-homekit #20: SparkFun ESP32 Thing

yene, 2018-05-19:
I got it working with the Adafruit HUZZAH32, now I am trying the SparkFun ESP32 Thing.

After flashing I only get garbled characters from the serial connection. An idea what the problem could be?

younghyunjo, 2018-05-21:
I am sorry I don't have any idea.

younghyunjo, 2018-05-21:
I use nodemcu

streicherlouw, 2018-05-21:
The sparkfun board uses a 26 MHz crystal in stead of the more usual 40MHz. Be sure to set the the option in your ESP dev environment with make menuconfig under esp32 specific options. I have used this code on both the sparkfun esp thing and a nodemcu, and needed to make this change.

streicherlouw, 2018-05-21:
It can be set in "make menuconfig" under "Component Config" -> ESP32-Specific -> Main XTAL frequency.

Choices are:
40 MHZ
26 MHz
Autodetect

yene, 2018-05-21:
@streicherlouw thank you it works

@younghyunjo I didn't know they exist, who is the manufacturer of your nodemcu?

younghyunjo, 2018-05-21:
@yene 
I bought it from [here](https://www.aliexpress.com/item/ESP32-ESP32S-Development-Board-2-4GHz-WiFi-Bluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32/32857439453.html?spm=2114.10010108.1000014.3.53204abb8oJRo9&scm=1007.13338.98644.000000000000000&scm_id=1007.13338.98644.000000000000000&scm-url=1007.13338.98644.000000000000000&pvid=38de04a6-ffe9-4142-b552-622d63da63c2&_t=pvid:38de04a6-ffe9-4142-b552-622d63da63c2,scm-url:1007.13338.98644.000000000000000).

closed 2018-05-21 by yene