younghyunjo/esp32-homekit #22: Button push for switch

givanildo, 2018-06-18:
dear younghyunjo !!! congratulations for yuor job!
sorry about my bad English....
May you help me  insert in the code a push button for physic(local PULL_UP) change the switch ?
i don't know how to do this.

wumbo, 2018-07-12:
Hi givanildo. I created a fork of this project with a touch switch example. It lets you use the ESP32 as a physical HomeKit switch to control automations, rather than turning on/off an LED on the device.

The idea is to call the lead_write function (I renamed to switch_write) when you detect a change on your input. In my case I used the touch sensor on GPIO4, but you could easily modify it to use a push button. Check it out [here](https://github.com/wumbo/esp32-homekit-touch-switch) in the `touch-switch` example folder

givanildo, 2018-07-12:
Thank you !   I appreciate your job!  Thanks for answer my question!

younghyunjo, 2018-07-18:
@wumbo  
Thanks for answer the question :+1:

closed 2018-07-18 by younghyunjo