younghyunjo/esp32-homekit #23: Never free

DigitalStealth, 2018-06-27:
https://github.com/younghyunjo/esp32-homekit/blob/f5acc6d6c8349c79ba1dec407b1968fe21e889a3/src/accessories.c#L320

Hey @younghyunjo! First of all I would like to thanks you for your perfect work! It is a simple way to use HAP for ESP32 with your library.

But I found a couple mistake which cause the memory leaks. In my case it happened very quickly because my accessory has 8 services with 2-3 characteristics for every service.

It this ticket `cJSON` object was created but newer free. You must to use `cJSON_Delete()` for free this instance of` cJSON` pointer.

Also I have a question/request for your library about advertise HAP accessory.
I mean the "sf" TXT record item. The documentation said that parameter must be equal to 1 if accessory has not been paired with any controller, but if accessory already has been paired this parameter must be a zero. In your code this parameter always is 1 because the value of `adv->state` is always `ADVERTISE_ACCESSORY_STATE_NOT_PAIRED` (I mean

`static void _service_txt_set(struct advertiser* adv);`

in advertise.c).
Could you please change your code for changing this parameter?

Regards

younghyunjo, 2018-06-28:
I leave the "sf" to 1 for test purpose.
If you want to change flag, you can use "advertise_accessory_state_set" api.

DigitalStealth, 2018-06-29:
Hey @younghyunjo, thanks for your answer!

I found this API method but how can I understand that pairing has been made?

younghyunjo, 2018-07-18:
@DigitalStealth 

Sorry for late reply
If pairing has been maed, the code hits esp32-homekit/src/pair_verify.c, line 209

closed 2018-07-19 by younghyunjo

DigitalStealth, 2018-07-19:
Hey @younghyunjo, thanks for your reply!