younghyunjo/esp32-homekit #26: How to distinguish different types of rooms

simonlou99, 2018-07-30:
I added several devices and found them all in the same room.
How to distinguish different types of rooms?
Tks.

cduvenhorst, 2018-07-30:
This is no issue with the server code. That‘s something the client side has to do.

If your client is the Apple home app, you are able to set the room in the accessory settings (deep press and settings).

simonlou99, 2018-07-30:
I did it as you said. When you set a room type for one device, all other devices are the same type  
Please kindly help check the code to fix the issue. Tks .
void hap_object_init(void* arg)
{
    void* accessory_object = hap_accessory_add(a);
    struct hap_characteristic cs[] = {
        {HAP_CHARACTER_IDENTIFY, (void*)true, NULL, identify_read, NULL, NULL},
        {HAP_CHARACTER_MANUFACTURER, (void*)MANUFACTURER_NAME, NULL, NULL, NULL, NULL},
        {HAP_CHARACTER_MODEL, (void*)MODEL_NAME, NULL, NULL, NULL, NULL},
        {HAP_CHARACTER_NAME, (void*)ACCESSORY_NAME, NULL, NULL, NULL, NULL},
        {HAP_CHARACTER_SERIAL_NUMBER, (void*)"0123456789", NULL, NULL, NULL, NULL},
        {HAP_CHARACTER_FIRMWARE_REVISION, (void*)"1.0", NULL, NULL, NULL, NULL},
    };
    hap_service_and_characteristics_add(a, accessory_object, HAP_SERVICE_ACCESSORY_INFORMATION, cs, ARRAY_SIZE(cs));
    struct hap_characteristic cc[] = {
        {HAP_CHARACTER_ON, (void*)led, NULL, led_read, led_write, led_notify},
    };
    hap_service_and_characteristics_add(a, accessory_object, HAP_SERVICE_SWITCHS, cc, ARRAY_SIZE(cc));

    struct hap_characteristic temperature_sensor[] = {
        {HAP_CHARACTER_CURRENT_TEMPERATURE, (void*)temp, NULL, temp_read, NULL, temp_notify},
        {HAP_CHARACTER_NAME, (void*)"Temp" , NULL, NULL, NULL, NULL},
        hap_service_and_characteristics_add(a , accessory_object, HAP_SERVICE_TEMPERATURE_SENSOR, temperature_sensor, ARRAY_SIZE(temperature_sensor)); 

 struct hap_characteristic humidity_sensor[] = {
        {HAP_CHARACTER_CURRENT_RELATIVE_HUMIDITY, (void*)humidity, NULL, _humidity_read, NULL, _humidity_notify},
        {HAP_CHARACTER_NAME, (void*)"HYGROMETER" , NULL, NULL, NULL, NULL},
    };
    hap_service_and_characteristics_add(a , accessory_object, HAP_SERVICE_HUMIDITY_SENSOR, humidity_sensor, ARRAY_SIZE(humidity_sensor));
  ...............

younghyunjo, 2018-07-31:
@simonlou99 

How about distinguish by name?

simonlou99, 2018-07-31:
I wanted to add a lot of devices, but I found them all in the same room type in the apple homekit app, and if I change one room type, the other devices change with it. The equipment is always in the same room.
You can add more than one device and test it out.

younghyunjo, 2018-07-31:
Do you want put every room sensons on single esp32?

younghyunjo, 2018-08-01:
You already did. 
You add many services with  characteristics at single esp32.