younghyunjo/esp32-homekit #30: iOS 12 beta signature failed

CMGeorge, 2018-09-03:
HAP: HTTP request from 192.168.1.241:54491: POST /pair-setup
[PAIR-SETUP] STATE:5tlv_decode failed. type:10
tlv length:50 0x320x01 0x24 0x46 0x35 0x34 0x43 0x44 0x36 0x42 0x41 0x2D 0x33 0x39 0x38 0x45 0x2D 0x34 0x36 0x33 0x42 0x2D 0x41 0x30 0x36 0x45 0x2D 0x35 0x35 0x35 0x34 0x36 0x36 0x43 0x44 0x45 0x46 0x37 0x46 0x03 0x20 0xBE 0x4B 0xBC 0xEA 0x47 0xC0 0xA0 0x4E 0x42 0x21 _io
s_device_signature_verify failed
[HTTPD] DEFAULT:102
[HTTPD] DEFAULT:100
[HTTPD] MG_EV_SEND. 0
[HTTPD] MG_EV_CLOSEConnection 0x3ffbbd18 closed

Any clue what's happen here?

kisgyorgy, 2018-09-24:
Same problem in my environment. 
Will there be any update soon?

JBL-IMT, 2018-10-11:
Dear All,
Same problem with the "official" ios12 release.
I think that the problem is on ligne 116 of pair_setup.c. (ios_device_signature).
But, i don't understand why ...

Thx,

pablomarx, 2018-10-17:
I believe the issue is the improper use of strlen() inside [_subtlv_decrypt](https://github.com/younghyunjo/esp32-homekit/blob/f5acc6d6c8349c79ba1dec407b1968fe21e889a3/src/pair_setup.c#L47).  

Changing:

`*subtlv_length = strlen((char*)*subtlv);` 

to: 

`*subtlv_length = encrypted_tlv->length;` 

resolves the issue for me.

Inside `_ios_device_signature_verify`, `tlv_decode` failed to find the subtlv of type 10 (`HAP_TLV_TYPE_SIGNATURE`).  Dumping the contents of subtlv showed there were only two embedded types: `HAP_TLV_TYPE_IDENTIFIER` and `HAP_TLV_TYPE_PUBLICKEY`.   The HAP documentation states all three items should exist – so the lack of one is suspicious. Further, in my environment the `HAP_TLV_TYPE_PUBLICKEY` was being truncated by 24 bytes (8 bytes existed, despite the sub-tlv claiming 32 bytes should be there).   This indicated the data was being truncated somewhere...

Given this truncation, I worked back to where  `_ios_device_signature_verify` is invoked: `_setup_m6`.  Seeing that `device_msg_length` was significantly larger than `device_subtlv_length`, this seemed to indicate some problem in `_subtlv_decrypt`.  Given a TLV could contain 0x00, it's use of strlen immediately seemed problematic.

JBL-IMT, 2018-10-18:
Dear all,

Fine ! 

I change *subtlv_length = strlen((char*)*subtlv); to: *subtlv_length = encrypted_tlv->length;

It's working for me (ios12.0)

I will try this week-end to investigate the problem in _subtlv_decrypt.

Sincerely,

sorriso93, 2018-11-08:
Hello, IOS12, changed the subtlv_length but no pairing... Home app see the switch but it hang pairing it for 3 minutes and then says it is not possible to add SWITCH...

VinceFilary, 2018-11-25:
Just paired to switch on iOS 12.1
Pairing code provided in README is WRONG!

Looked at main.c, pairing code is actually:    053-58-197
(README has the last three as 917)

ALSO, if you have made the changed suggested in this thread, changing:
*subtlv_length = strlen((char*)*subtlv); 
to 
*subtlv_length = encrypted_tlv->length;

it will also NOT work.

Had to change back to original code, then paired fine.