younghyunjo/esp32-homekit #42: Rebooting...

ruslangirichev, 2019-03-23:
Hello, 
I've tried this with several devices but always have the same problem — it is rebooting infinitely. Do I guess right it should not do that?

`Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6332
load:0x40078000,len:11308
ho 0 tail 12 room 4
load:0x40080400,len:6680
entry 0x40080760
I (31) boot: ESP-IDF v4.0-dev-141-g106dc0590 2nd stage bootloader
I (31) boot: compile time 22:02:04
I (31) boot: Enabling RNG early entropy source...
I (37) boot: SPI Speed      : 40MHz
I (41) boot: SPI Mode       : DIO
I (45) boot: SPI Flash Size : 4MB
I (49) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (75) boot:  2 factory          factory app      00 00 00010000 00100000
I (82) boot: End of partition table
I (87) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x24738 (149304) map
I (148) esp_image: segment 1: paddr=0x00034760 vaddr=0x3ffb0000 size=0x02de0 ( 11744) load
I (153) esp_image: segment 2: paddr=0x00037548 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/ruslan/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (155) esp_image: segment 3: paddr=0x00037950 vaddr=0x40080400 size=0x086c0 ( 34496) load
I (178) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x8eb60 (584544) map
0x400d0018: _stext at ??:?

I (383) esp_image: segment 5: paddr=0x000ceb80 vaddr=0x40088ac0 size=0x07090 ( 28816) load
0x40088ac0: ieee80211_output_process at ??:?

I (404) boot: Loaded app from partition at offset 0x10000
I (404) boot: Disabling RNG early entropy source...
I (405) cpu_start: Pro cpu up.
I (408) cpu_start: Application information:
I (413) cpu_start: Project name:     app-template
I (419) cpu_start: App version:      v1.0.1-21-gfb7392d-dirty
I (425) cpu_start: Compile time:     Mar 23 2019 22:02:09
I (431) cpu_start: ELF file SHA256:  6dee4d5824a56468...
I (437) cpu_start: ESP-IDF:          v4.0-dev-141-g106dc0590
I (443) cpu_start: Starting app cpu, entry point is 0x40081064
0x40081064: call_start_cpu1 at /Users/ruslan/esp/esp-idf/components/esp32/cpu_start.c:267

I (435) cpu_start: App cpu up.
I (454) heap_init: Initializing. RAM available for dynamic allocation:
I (461) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (467) heap_init: At 3FFB94D8 len 00026B28 (154 KiB): DRAM
I (473) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (480) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (486) heap_init: At 4008FB50 len 000104B0 (65 KiB): IRAM
I (492) cpu_start: Pro cpu start user code
I (175) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (340) wifi: wifi driver task: 3ffc1190, prio:23, stack:3584, core=0
I (340) wifi: wifi firmware version: 9c8361c
I (340) wifi: config NVS flash: enabled
I (340) wifi: config nano formating: disabled
I (350) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (360) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (390) wifi: Init dynamic tx buffer num: 32
I (390) wifi: Init data frame dynamic rx buffer num: 32
I (390) wifi: Init management frame dynamic rx buffer num: 32
I (390) wifi: Init management short buffer num: 32
I (400) wifi: Init static rx buffer size: 1600
I (400) wifi: Init static rx buffer num: 10
I (410) wifi: Init dynamic rx buffer num: 32
I (510) phy: phy_version: 4100, 6fa5e27, Jan 25 2019, 17:02:06, 0, 0
I (510) wifi: mode : sta (24:0a:c4:0e:4e:20)
I (510) SWITCH: wifi_init_sta finished.
I (510) SWITCH: connect to ap SSID:Hello password:810450970140
I (630) wifi: new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (630) wifi: state: init -> auth (b0)
I (630) wifi: state: auth -> assoc (0)
I (640) wifi: state: assoc -> run (10)
I (660) wifi: connected with Hello, channel 1, bssid = 74:1b:b2:d4:91:46
I (660) wifi: pm start, type: 1

I (5290) event: sta ip: 192.168.100.9, mask: 255.255.255.0, gw: 192.168.100.1
I (5290) SWITCH: got ip:192.168.100.9
E (5290) nvs: nothing saved. key:240AC40E4E20D0

E (5300) nvs: nothing saved. key:240AC40E4E20D1

E (5300) nvs: nothing saved. key:240AC40E4E20D2

E (5310) nvs: nothing saved. key:240AC40E4E20D3

E (5310) nvs: nothing saved. key:240AC40E4E20D4

E (5320) nvs: nothing saved. key:240AC40E4E20D5

E (5320) nvs: nothing saved. key:240AC40E4E20D6

E (5330) nvs: nothing saved. key:240AC40E4E20D7

/Users/ruslan/esp/esp-idf/components/freertos/queue.c:729 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008ca53 on core 0
0x4008ca53: xQueueGenericSend at /Users/ruslan/esp/esp-idf/components/freertos/queue.c:2044

ELF file SHA256: 6dee4d5824a56468a395017c24f409093a91ae9365e49a36f41e5eafbc615cdc

Backtrace: 0x40086fb8:0x3ffbf7a0 0x40087205:0x3ffbf7c0 0x4008ca53:0x3ffbf7e0 0x4011f429:0x3ffbf820 0x4011f235:0x3ffbf850 0x400d2c2a:0x3ffbf8e0 0x401364ad:0x3ffbf940 0x401364f1:0x3ffbf960 0x4008c4bd:0x3ffbf9b0
0x40086fb8: invoke_abort at /Users/ruslan/esp/esp-idf/components/esp32/panic.c:715

0x40087205: abort at /Users/ruslan/esp/esp-idf/components/esp32/panic.c:715

0x4008ca53: xQueueGenericSend at /Users/ruslan/esp/esp-idf/components/freertos/queue.c:2044

0x4011f429: httpd_bind at /Users/ruslan/esp/esp32-homekit/src/httpd.c:83

0x4011f235: hap_accessory_register at /Users/ruslan/esp/esp32-homekit/src/hap.c:470

0x400d2c2a: event_handler at /Users/ruslan/esp/esp32-homekit/examples/switch/main/main.c:128 (discriminator 9)

0x401364ad: esp_event_post_to_user at /Users/ruslan/esp/esp-idf/components/esp_event/event_loop.c:109

0x401364f1: esp_event_loop_task at /Users/ruslan/esp/esp-idf/components/esp_event/event_loop.c:109

0x4008c4bd: vPortTaskWrapper at /Users/ruslan/esp/esp-idf/components/freertos/port.c:403

Rebooting...`

COBECTb, 2019-04-10:
+

mpietron, 2019-04-27:
same problem

willson556, 2019-05-02:
I've had somewhat better luck with ESP-IDF v3.1 however it does seem to lock up after many hours -- still investigating the root cause.

willson556, 2019-05-06:
I've made several reliability (and minor feature) improvements to the library at [willson556/esp32-homekit](https://github.com/willson556/esp32-homekit) and it seems to be stable running on ESP-IDF v3.1.

jussiedlund, 2019-05-11:
I had the same issue with ESP-IDF v3.1. It worked after I switched to ESP-IDF v3.2.

closed 2019-05-14 by ruslangirichev