younghyunjo/esp32-homekit #44: NVS: Nothing saved. key:xxxxxxxxxxxx

baaskaas, 2019-05-30:
Hello, trying out the basic switch tutorial before results in this error. 
I just eddited my wifi credentials. As you can see it is connected. Right after that the system fails. Maybe i got the wrong package of something?
maybe a wrong setting in the config?

I don't seem to get whats wrong. 

here is the log:

`
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6380
load:0x40078000,len:11304
load:0x40080400,len:6700
entry 0x40080764
I (29) boot: ESP-IDF v4.0-dev-667-gda13efc17 2nd stage bootloader
I (29) boot: compile time 17:43:10
I (29) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x24598 (148888) map
I (146) esp_image: segment 1: paddr=0x000345c0 vaddr=0x3ffb0000 size=0x0306c ( 12396) load
I (151) esp_image: segment 2: paddr=0x00037634 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/user/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (153) esp_image: segment 3: paddr=0x00037a3c vaddr=0x40080400 size=0x085d4 ( 34260) load
I (176) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x8e734 (583476) map
0x400d0018: _stext at ??:?

I (380) esp_image: segment 5: paddr=0x000ce754 vaddr=0x400889d4 size=0x072bc ( 29372) load
0x400889d4: ppSearchTxQueue at ??:?

I (402) boot: Loaded app from partition at offset 0x10000
I (402) boot: Disabling RNG early entropy source...
I (403) cpu_start: Pro cpu up.
I (406) cpu_start: Application information:
I (411) cpu_start: Project name:     app-template
I (417) cpu_start: App version:      v1.0.1-21-gfb7392d-dirty
I (423) cpu_start: Compile time:     May 30 2019 17:43:12
I (429) cpu_start: ELF file SHA256:  40afeca62a0ea8fe...
I (435) cpu_start: ESP-IDF:          v4.0-dev-667-gda13efc17
I (441) cpu_start: Starting app cpu, entry point is 0x4008105c
0x4008105c: call_start_cpu1 at /Users/user/esp/esp-idf/components/esp32/cpu_start.c:267

I (434) cpu_start: App cpu up.
I (452) heap_init: Initializing. RAM available for dynamic allocation:
I (459) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (465) heap_init: At 3FFB96C8 len 00026938 (154 KiB): DRAM
I (471) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (478) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (484) heap_init: At 4008FC90 len 00010370 (64 KiB): IRAM
I (490) cpu_start: Pro cpu start user code
I (173) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (273) wifi: wifi driver task: 3ffc10c8, prio:23, stack:3584, core=0
I (273) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (273) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
I (303) wifi: wifi firmware version: e11ccc4
I (303) wifi: config NVS flash: enabled
I (303) wifi: config nano formating: disabled
I (303) wifi: Init dynamic tx buffer num: 32
I (303) wifi: Init data frame dynamic rx buffer num: 32
I (313) wifi: Init management frame dynamic rx buffer num: 32
I (313) wifi: Init management short buffer num: 32
I (323) wifi: Init static rx buffer size: 1600
I (323) wifi: Init static rx buffer num: 10
I (333) wifi: Init dynamic rx buffer num: 32
I (433) phy: phy_version: 4100, 2a5dd04, Jan 23 2019, 21:00:07, 0, 0
I (433) wifi: mode : sta (30:ae:a4:0a:6c:5c)
I (433) SWITCH: wifi_init_sta finished.
I (443) SWITCH: connect to ap SSID:TjeppieNet password:boem1234
I (803) wifi: new:<13,0>, old:<1,0>, ap:<255,255>, sta:<13,0>, prof:1
I (803) wifi: state: init -> auth (b0)
I (803) wifi: state: auth -> assoc (0)
I (813) wifi: state: assoc -> run (10)
I (853) wifi: connected with TjeppieNet, channel 13, bssid = c4:ea:1d:e1:c2:bf
I (853) wifi: pm start, type: 1

I (1753) tcpip_adapter: sta ip: 192.168.1.98, mask: 255.255.255.0, gw: 192.168.1.1
I (1753) SWITCH: got ip:192.168.1.98
E (1753) nvs: nothing saved. key:30AEA40A6C5CD0

E (1763) nvs: nothing saved. key:30AEA40A6C5CD1

E (1763) nvs: nothing saved. key:30AEA40A6C5CD2

E (1773) nvs: nothing saved. key:30AEA40A6C5CD3

E (1773) nvs: nothing saved. key:30AEA40A6C5CD4

E (1783) nvs: nothing saved. key:30AEA40A6C5CD5

E (1783) nvs: nothing saved. key:30AEA40A6C5CD6

E (1793) nvs: nothing saved. key:30AEA40A6C5CD7

/Users/user/esp/esp-idf/components/freertos/queue.c:729 (xQueueGenericSend)- assert failed!
abort() was called at PC 0x4008bfa7 on core 0
0x4008bfa7: xQueueGenericSend at /Users/user/esp/esp-idf/components/freertos/queue.c:2044

ELF file SHA256: 40afeca62a0ea8fe77c03ebc9afc0c8d4178b9d75d5dfeb05e9b1d8b9df98685

Backtrace: 0x40087b94:0x3ffbf520 0x40087de1:0x3ffbf540 0x4008bfa7:0x3ffbf560 0x400e9dc9:0x3ffbf5a0 0x400e9bed:0x3ffbf5d0 0x400d33d2:0x3ffbf660 0x401008d9:0x3ffbf6c0 0x4015d6aa:0x3ffbf6e0 0x40101131:0x3ffbf710 0x401011f1:0x3ffbf750 0x4008ba15:0x3ffbf770
0x40087b94: invoke_abort at /Users/user/esp/esp-idf/components/esp32/panic.c:716

0x40087de1: abort at /Users/user/esp/esp-idf/components/esp32/panic.c:716

0x4008bfa7: xQueueGenericSend at /Users/user/esp/esp-idf/components/freertos/queue.c:2044

0x400e9dc9: httpd_bind at /Users/user/esp/esp32-homekit/src/httpd.c:83

0x400e9bed: hap_accessory_register at /Users/user/esp/esp32-homekit/src/hap.c:470

0x400d33d2: event_handler at /Users/user/esp/esp32-homekit/examples/switch/main/main.c:128 (discriminator 9)

0x401008d9: esp_event_post_to_user at /Users/user/esp/esp-idf/components/esp_event/event_loop_legacy.c:44

0x4015d6aa: handler_execute at /Users/user/esp/esp-idf/components/esp_event/esp_event.c:145

0x40101131: esp_event_loop_run at /Users/user/esp/esp-idf/components/esp_event/esp_event.c:641

0x401011f1: esp_event_loop_run_task at /Users/user/esp/esp-idf/components/esp_event/esp_event.c:641

0x4008ba15: vPortTaskWrapper at /Users/user/esp/esp-idf/components/freertos/port.c:403

Rebooting...
ets Jun  8 2016 00:22:57
`

baaskaas, 2019-05-31:
Lol Fixed it. 
I was on ESP-IDF v4.0-dev-667-gda13efc17
Now on:
v3.2
thats what it said. 
i downgraded by removing the esp-idf folder and then following this guide to get the esp-idf from github
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/index.html#get-started-get-esp-idf

On MacOS 10.14.5

pranay-crypto, 2019-06-29:
/home/pranay/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:155:44: error: expected initializer before '__result_use_check'
 void *reallocarray(void *, size_t, size_t) __result_use_check __alloc_size(2)
                                            ^~~~~~~~~~~~~~~~~~
/home/pranay/esp/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:340:52: error: expected initializer before '__alloc_align'
 void * aligned_alloc(size_t, size_t) __malloc_like __alloc_align(1)
                                                    ^~~~~~~~~~~~~
/home/pranay/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'asio/asio/src/asio.o' failed
make[1]: *** [asio/asio/src/asio.o] Error 1
/home/pranay/esp/esp-idf/make/project.mk:530: recipe for target 'component-asio-build' failed
make: *** [component-asio-build] Error 2

This error occurs while make after downgrading esp-idf now what to do

Xanadu1220, 2019-06-30:
Have the same question 😭😭😭 How to solve it?

pranay-crypto, 2019-06-30:
This is because of following two guides at same time both esp-idf guide have different process of installation so , Delete all your previous installed file and start again by following this guide given below :

https://docs.espressif.com/projects/esp-idf/en/stable/get-started/linux-setup.html

After that error is fixed

Xanadu1220, 2019-06-30:
This is so weird QAQ 
After it, still come out !!!

Xanadu1220, 2019-06-30:
what should I config of main.c ? Just #if 1
#define EXAMPLE_ESP_WIFI_SSID "XXX"
#define EXAMPLE_ESP_WIFI_PASS "XXX"
???

pranay-crypto, 2019-06-30:
Write your WiFi router name (ssid) and password in the place of XXX

Xanadu1220, 2019-07-01:
Thanks bro ! I Done !!!!

rillhu, 2020-04-09:
you should select the right ESP-IDF version. you can also refer to https://github.com/rillhu/esp32-homekit, which I made some update based on this repo.

jondthompson, 2020-05-10:
What would be needed to update the project for esp-idf4?