younghyunjo/esp32-homekit #47: Failed to compile

Benji1992-HU, 2019-08-08:
Hy!
I tryed to complie the example project(switch) with Cmake, but I have the following error:

esp32/libesp32.a(cpu_start.c.obj):(.literal.main_task+0x18): undefined reference to `app_main'
esp32/libesp32.a(cpu_start.c.obj): In function `main_task':
C:/esp-idf/components/esp32/cpu_start.c:489: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

The full log:
C:\esp-idf\examples\esp32-homekit\examples\switch>idf.py build
Note: You are using Python 3.7.3. Python 3 support is new, please report any problems you encounter. Search for 'Setting the Python Interpreter' in the ESP-IDF docs if you want to use Python 2.7.
Checking Python dependencies...
Python requirements from C:\esp-idf\requirements.txt are satisfied.
Running ninja in directory C:\esp-idf\examples\esp32-homekit\examples\switch\build
Executing "ninja all"...
[0/1] Re-running CMake...
CMake Warning at C:/esp-idf/tools/cmake/scripts/expand_requirements.cmake:141 (message):
  Component C:/esp-idf/examples/esp32-homekit/examples/switch/main contains
  old-style component.mk but no CMakeLists.txt.  Component will be skipped.
Call Stack (most recent call first):
  C:/esp-idf/tools/cmake/scripts/expand_requirements.cmake:266 (components_find_all)

-- Building empty aws_iot component due to configuration
-- Component names: soc log heap freertos vfs newlib esp_ringbuf driver esp_event ethernet lwip tcpip_adapter partition_table app_update spi_flash mbedtls micro-ecc bootloader_support nvs_flash pthread smartconfig_ack wpa_supplicant xtensa-debug-module esp32 cxx app_trace asio jsmn aws_iot bootloader bt coap console nghttp esp-tls esp_adc_cal tcp_transport esp_http_client esp_http_server esp_https_ota esptool_py expat wear_levelling sdmmc fatfs freemodbus idf_test json libsodium mdns mqtt openssl protobuf-c protocomm spiffs ulp wifi_provisioning
-- Component paths: C:/esp-idf/components/soc;C:/esp-idf/components/log;C:/esp-idf/components/heap;C:/esp-idf/components/freertos;C:/esp-idf/components/vfs;C:/esp-idf/components/newlib;C:/esp-idf/components/esp_ringbuf;C:/esp-idf/components/driver;C:/esp-idf/components/esp_event;C:/esp-idf/components/ethernet;C:/esp-idf/components/lwip;C:/esp-idf/components/tcpip_adapter;C:/esp-idf/components/partition_table;C:/esp-idf/components/app_update;C:/esp-idf/components/spi_flash;C:/esp-idf/components/mbedtls;C:/esp-idf/components/micro-ecc;C:/esp-idf/components/bootloader_support;C:/esp-idf/components/nvs_flash;C:/esp-idf/components/pthread;C:/esp-idf/components/smartconfig_ack;C:/esp-idf/components/wpa_supplicant;C:/esp-idf/components/xtensa-debug-module;C:/esp-idf/components/esp32;C:/esp-idf/components/cxx;C:/esp-idf/components/app_trace;C:/esp-idf/components/asio;C:/esp-idf/components/jsmn;C:/esp-idf/components/aws_iot;C:/esp-idf/components/bootloader;C:/esp-idf/components/bt;C:/esp-idf/components/coap;C:/esp-idf/components/console;C:/esp-idf/components/nghttp;C:/esp-idf/components/esp-tls;C:/esp-idf/components/esp_adc_cal;C:/esp-idf/components/tcp_transport;C:/esp-idf/components/esp_http_client;C:/esp-idf/components/esp_http_server;C:/esp-idf/components/esp_https_ota;C:/esp-idf/components/esptool_py;C:/esp-idf/components/expat;C:/esp-idf/components/wear_levelling;C:/esp-idf/components/sdmmc;C:/esp-idf/components/fatfs;C:/esp-idf/components/freemodbus;C:/esp-idf/components/idf_test;C:/esp-idf/components/json;C:/esp-idf/components/libsodium;C:/esp-idf/components/mdns;C:/esp-idf/components/mqtt;C:/esp-idf/components/openssl;C:/esp-idf/components/protobuf-c;C:/esp-idf/components/protocomm;C:/esp-idf/components/spiffs;C:/esp-idf/components/ulp;C:/esp-idf/components/wifi_provisioning
-- ccache will be used for faster builds
-- Adding linker script C:/esp-idf/examples/esp32-homekit/examples/switch/build/esp32/esp32_out.ld
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.common.ld
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.rom.ld
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.peripherals.ld
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script C:/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld
-- Building empty aws_iot component due to configuration
-- Component libraries: soc;log;heap;freertos;vfs;newlib;esp_ringbuf;driver;esp_event;ethernet;lwip;tcpip_adapter;app_update;spi_flash;mbedtls;micro-ecc;bootloader_support;nvs_flash;pthread;smartconfig_ack;wpa_supplicant;xtensa-debug-module;esp32;cxx;app_trace;asio;jsmn;aws_iot;bt;coap;console;nghttp;esp-tls;esp_adc_cal;tcp_transport;esp_http_client;esp_http_server;esp_https_ota;expat;wear_levelling;sdmmc;fatfs;freemodbus;idf_test;json;libsodium;mdns;mqtt;openssl;protobuf-c;protocomm;spiffs;ulp;wifi_provisioning
-- Configuring done
-- Generating done
-- Build files have been written to: C:/esp-idf/examples/esp32-homekit/examples/switch/build
[1/5] Performing build step for 'bootloader'
ninja: no work to do.
[2/3] Linking CXX executable esp32-homekit.elf
FAILED: esp32-homekit.elf
cmd.exe /C "cd . && C:\Users\Benji\.espressif\tools\xtensa-esp32-elf\1.22.0-80-g6c4433a5-5.2.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe    -nostdlib CMakeFiles/esp32-homekit.elf.dir/dummy_main_src.c.obj  -o esp32-homekit.elf  -Wl,--gc-sections -Wl,--cref -Wl,--Map=esp32-homekit.map -Wl,--start-group soc/libsoc.a log/liblog.a heap/libheap.a freertos/libfreertos.a vfs/libvfs.a newlib/libnewlib.a esp_ringbuf/libesp_ringbuf.a driver/libdriver.a esp_event/libesp_event.a ethernet/libethernet.a lwip/liblwip.a tcpip_adapter/libtcpip_adapter.a app_update/libapp_update.a spi_flash/libspi_flash.a mbedtls/libmbedtls.a micro-ecc/libmicro-ecc.a bootloader_support/libbootloader_support.a nvs_flash/libnvs_flash.a pthread/libpthread.a smartconfig_ack/libsmartconfig_ack.a wpa_supplicant/libwpa_supplicant.a xtensa-debug-module/libxtensa-debug-module.a esp32/libesp32.a cxx/libcxx.a app_trace/libapp_trace.a asio/libasio.a jsmn/libjsmn.a coap/libcoap.a console/libconsole.a nghttp/libnghttp.a esp-tls/libesp-tls.a esp_adc_cal/libesp_adc_cal.a tcp_transport/libtcp_transport.a esp_http_client/libesp_http_client.a esp_http_server/libesp_http_server.a esp_https_ota/libesp_https_ota.a expat/libexpat.a wear_levelling/libwear_levelling.a sdmmc/libsdmmc.a fatfs/libfatfs.a freemodbus/libfreemodbus.a json/libjson.a libsodium/liblibsodium.a mdns/libmdns.a mqtt/libmqtt.a openssl/libopenssl.a protobuf-c/libprotobuf-c.a protocomm/libprotocomm.a spiffs/libspiffs.a ulp/libulp.a wifi_provisioning/libwifi_provisioning.a -Wl,--undefined=uxTopUsedPriority -L C:/esp-idf/components/newlib/lib -lc -lm -L C:/esp-idf/components/esp32/lib -L C:/esp-idf/components/esp32/ld/wifi_iram_opt -lcoexist -lcore -lespnow -lmesh -lnet80211 -lphy -lpp -lrtc -lsmartconfig -lwpa2 -lwpa -lwps -L C:/esp-idf/examples/esp32-homekit/examples/switch/build/esp32 -T esp32_out.ld -L C:/esp-idf/components/esp32/ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld -T esp32.rom.libgcc.ld -T esp32.rom.spiram_incompatible_fns.ld C:/esp-idf/components/esp32/libhal.a -lgcc -u call_user_start_cpu0 -u ld_include_panic_highint_hdl -lstdc++ -u __cxa_guard_dummy -u __cxx_fatal_exception -lgcov && cd ."
esp32/libesp32.a(cpu_start.c.obj):(.literal.main_task+0x18): undefined reference to `app_main'
esp32/libesp32.a(cpu_start.c.obj): In function `main_task':
C:/esp-idf/components/esp32/cpu_start.c:489: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

Any idea to solve thsi problem?

jangolen, 2019-08-17:
Try to compile with make in old style from msys2. https://docs.espressif.com/projects/esp-idf/en/latest/get-started-legacy/windows-setup.html

rillhu, 2020-04-09:
please refer to https://github.com/rillhu/esp32-homekit, which I made some update based on this repo.